yoneken/rosserial_stm32 #4: Trying to use your code

elgarbe, 2018-04-13:
Hi, I was trying this:
https://answers.ros.org/question/277324/stm32-with-ros-rosserial-in-eclipse-how/
and make it working ok, but now I found your solution and seems to be very nice.

Could you explain a little more how to start using it?
I'm starting with CubeMX project, with 2 GPIO and an UART configured. What is the next step?

Thank

yoneken, 2018-06-01:
HI!
Thank you for your comment, and sorry for my late reply.
This rosserial_stm32 is a part project of [rosserial](http://wiki.ros.org/rosserial). So, please read it first and install rosserial.
Then, please read a [readme of rosserial_stm32](https://github.com/yoneken/rosserial_stm32/blob/master/README.md) and [examples](https://github.com/yoneken/rosserial_stm32/tree/master/src/ros_lib/examples).

If you get another issue, please feel free to ask me!

elgarbe, 2018-07-02:
Ok, I've started over from scratch.
I clone rosserial from github (https://github.com/ros-drivers/rosserial), into a new catkin_ws. Run catkin_make, matking_make install and source devel/setup.bash
Now, what is the next step? do I need to clone rosserial_stm32 into where?

elgarbe, 2018-07-02:
I just tried to clone https://github.com/yoneken/rosserial_stm32.git into my git folder, then create a symlink into /home/elgarbe/catkin_ws_rosserial/src/rosserial/, then run catkin_make, catkin_make install and source devel/setup.bash.
Then create a ws: mkdir /home/elgarbe/ws_rosserial and cd to it
finally run rosrun rosserial_stm32 make_libraries.py .
and get this error:
```
Exporting to .
Traceback (most recent call last):
  File "/home/elgarbe/catkin_ws_rosserial/src/rosserial/rosserial_stm32/src/rosserial_stm32/make_libraries.py", line 90, in <module>
    shutil.copy(rosserial_stm32_dir+"/src/ros_lib/"+f, path+"/Inc/")
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 21] Is a directory: './Inc/'
```

Supgb, 2018-07-29:
you need to create a stm32 project with SW4STM32, and this will create a directory contain some folders(src, inc, ...). Finally change "Inc" to "inc" in make_libraries.py.

elgarbe, 2018-08-09:
Ok, I got it.
Could you tell me how fast a topic could be publish? I want to publish an IMU msg at 20Hz. Coult it be possible?

yoneken, 2018-08-09:
> Could you tell me how fast a topic could be publish?

It depends on your mpu, cpu, and occupancy of USB bus.
Let's try on-site! Basically, 20Hz is enough slow frequency.

elgarbe, 2018-08-09:
have you got an example using USB port?
I'm using your code on a STM32F722
If I create a project with USB with CubeMX I can't make it C++

yoneken, 2018-08-09:
> have you got an example using USB port?

Sorry. I don't have the experience to use USB port on STM32. If you cannot change the project to C++, it will be the devil to use rosserial. Because the original rosserial code is based on C++.

elgarbe, 2018-08-10:
if I use a CubeMX proyect with USB enabled then I can change the project to C++, but it won't compile.
I will stick with uart.
I'm using your code, but I get a lot of "Lost sync with device". What could be the problem? I'm just sending a string msg...

yoneken, 2018-08-13:
According to [this great advice](https://answers.ros.org/question/11237/rosserial-lost-sync-with-device/), you may forget to call spinOnce()?
In my sample code, I call it [on the main loop](https://github.com/yoneken/rosserial_stm32/blob/master/src/ros_lib/examples/chatter/Src/mainpp.cpp#L38).

closed 2019-01-07 by yoneken