yoneken/rosserial_stm32 #8: Which part of "STM32Hardware.h" should I edit?

hogepiyo334, 2019-03-28:
# Japanese
私はSTM32 Nucleo L4シリーズを使おうとしているのですが，変更する箇所はインクルードするファイルのf3をl4に変えるだけでいいのですか？また，ほかに変更しなければいけないと考えられる箇所はありますか？

# English(machine translation)
I am trying to use the STM32 Nucleo L4 series, but is it only necessary to change f3 of the included file to l4? Also, are there any other places that you think you should change?

hogepiyo334, 2019-03-28:
その後，自分で探した結果自己解決しました．以下，修正内容．
STM32Hardware.hのf3をl4に変える．
chatterの場合はCubeMXでDMA(RX=Circlar,TX=Nomal),USART2(baud=57600,GPIO SettingsのUserLabelをRX/TXそれぞれVCP_RX/VCP_TX),Tim2(clock source=internalclock)に設定．

yoneken, 2019-04-01:
情報ありがとうございます！
F3以外のシリーズのサンプルも欲しいところなので、もし気が向いたらサンプルをPRいｔだければ幸いです。

closed 2019-04-01 by yoneken

hogepiyo334, 2019-04-02:
これからF4にも適用するつもりなのでL4と同じ方法で適用できたらやり方をまとめるつもりです。

blacktrow, 2023-03-10:
@yoneken @Nyarn-WTF 

I have stm32f407 discovery board and when I change edit the STM32Hardware.h to f4xx series board I get a couple of errors in the build that goes:
https://imgur.com/5gPEvVW

and I also get an syntax error on the class that's declared in the file:
https://imgur.com/IfuyiBE
(I checked if it's compiled in cpp and yes it does)

any idea how do I fix them? why does it happen?