yoneken/rosserial_stm32 #9: Unable to sync with device error when porting to STM32F7 Discovery

InterestingWalrus, 2019-04-10:
Hi, I'm trying to use your code and trying to adapt to the F7 Discovery board.    As I'm a Keil guy, I've tried to port it to UVision using the steps found in this link: https://blog.csdn.net/qq_37416258/article/details/84844051. I believe the writer is using your repository as a base. It does compile but I get the dreaded "(Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino" when I run rossserial. I've tried multiple UART ports and the issue is the same. Could you possibly help?

yoneken, 2019-04-16:
I may know such like issue "(Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python)". 
When I use several ROS topics, I have to [enlarge the trasmit buffer](https://github.com/yoneken/rosserial_stm32/blob/master/src/ros_lib/STM32Hardware.h#L55) . Please check it.

yoneken, 2020-01-16:
I close this issue. Because it seems to be resolved.

closed 2020-01-16 by yoneken