yoneken/rosserial_stm32 #10: Trying with STM32CubeIDE

frafrakki, 2019-08-08:
I'm trying to use your code as ST has developed new IDE which is compatible on C++ coding.
And at the time on building, I've encountered the error below.
undefined reference to `ros::normalizeSecNSec(unsigned long&, unsigned long&)'
Is it possible helping me fix this problem?

BTW, I'm using ros kinetic and NUCLEO-F303K8.

yoneken, 2019-08-22:
I'm so sorry for my late reply..
Of course, I'm interested in ST's new IDE, but currently I've not use it yet..
Maybe, It needs a time to fix..

fdila, 2019-10-01:
Hi! I've edited this code to work with STM32CubeIDE.
My repo is based on stm32f7, but it should be easy enough to adapt the hardware for your nucleo board

[Here's my repo](https://github.com/fdila/rosserial_stm32f7)

yoneken, 2019-10-03:
Great job!!

yoneken, 2020-01-14:
I tried STM32CubeIDE and encountered the same error.
> undefined reference to `ros::normalizeSecNSec(unsigned long&, unsigned long&)'

I realized that the SW4STM32 treats /Inc as a source folder but the STM32CubeIDE does not.
The difference creates the undefined reference error of normalizeSecNSec, because the function is declared at time.cpp which is located in /Inc directory.

If you add /Inc directory as a "Source Location" in the project property, the error can be fixed.
I am still thinking about how to fix the error in the code.

yoneken, 2020-01-14:
I add a comment for STM32CubeIDE users in README.

closed 2020-01-14 by yoneken

CodeOn-ArK, 2021-02-24:
> I tried STM32CubeIDE and encountered the same error.
>
> I realized that the SW4STM32 treats /Inc as a source folder but the STM32CubeIDE does not.
> The difference creates the undefined reference error [...]

Ho there! Thanks for this reference that u made. Will u help me out in one thing?
I have made all .h files inside Inc folder and C files in another folder. And included both of them in Source location but it still doesn't works!! Can u help me out! Will be very much thankful.

commit c4239dcf (2020-01-14): Add support for STM32CubeIDE. fix #10
commit 6f914dcd (2020-01-14): Add support for STM32CubeIDE. fix #10