yoneken/rosserial_stm32 #19: which version of rosserial do I need to use?

elgarbe, 2020-03-18:
Hi, I'm trying your repo, I've created a CubeMX project with just an uart3, with DMA in CIRCULAR mode, clone your repo, catkin_make it, cd to my project, run make_lib, and all other stuff. 
Project compile ok (i've change uart2 to 3 and define f7 uC). I've run it and on my computer run rosserial_server, but I get this WARN and there is not new topic (I'm trying chatter project):

[ WARN] [1584502199.794725206]: Bad message header length checksum. Dropping message from client. T41508 L29184 C239 114

elgarbe, 2020-03-18:
I'm sorry I made a mistake, I've configured both DMA in circular mode, but I should config just RX on circular mode. Now it is working ok.
I've receive this message:

```
odroid@odroid:~$ rosrun rosserial_server serial_node _port:=/dev/ttySAC0 _baud:=57600
[ INFO] [1584546346.221603568]: rosserial_server session configured for /dev/ttySAC0 at 57600bps.
[ INFO] [1584546346.222404701]: Opened /dev/ttySAC0
[ INFO] [1584546346.419244218]: waitForService: Service [/message_info] has not been advertised, waiting...
[ WARN] [1584546351.430242624]: Timed out waiting for message_info service to become available.
[ WARN] [1584546351.431878432]: Failed to call message_info service. Proceeding without full message definition.
[ WARN] [1584546351.432038600]: Advertising on topic [/chatter] with an empty message definition.  Some tools (e.g. rosbag) may not work correctly.
```

Is there a way to solve this issue?

elgarbe, 2020-03-18:
I'm using rosserial_python and it works great!
```

odroid@odroid:~$ rosrun rosserial_python serial_node.py _port:=/dev/ttySAC0  _baud:=57600
[INFO] [1584546496.899351]: ROS Serial Python Node
[INFO] [1584546496.911662]: Connecting to /dev/ttySAC0 at 57600 baud
[INFO] [1584546499.581059]: Note: publish buffer size is 512 bytes
[INFO] [1584546499.581864]: Setup publisher on chatter [std_msgs/String]
```
I will start testing your implementation on my final application and comment back. Good work!

elgarbe, 2020-03-18:
Finally!!!!!!! A really stable implementation of rosserial for STM32!!!!!
It works great. you can close this issue.
I have some doubt about your code and how to adapt to my use-case. Could I mail to you in private or do I need to raise an Issue, here? My email is lgarberoglio@gmail.com

Thank

elgarbe, 2020-03-21:
Hi, I would like to let you know that I've tested your code publishing tf and odom messages @20Hz. In order to do that I've changed buffer size to 1024 and uart speed to 921600. I'm using an F722 uC at 216 MHz.

yoneken, 2020-03-25:
I'm sorry for the late reply.
I'm happy to hear about your great result!
Have you already solved your issues? I can help you with mail in private. But if you open an issue here, it should help someone :-)

elgarbe, 2020-03-25:
Oh, no problem. Yes, it is working great, I've made a mistake with UART DMA in circular mode, I've config both tx and RX in circualr mode, but just RX has to be in circular mode.
Then there was some error with roserial_server, but I've changed to rosserial_python and everithing start working.
Then I've change UART speed and Buffer size in order to get more publishers and more publish rate. It is working really good.

Thank

closed 2020-03-25 by yoneken