yoneken/rosserial_stm32 #21: Topics are not being seen by ROS

Abo3abed, 2020-03-28:
Hey! 
I'm using your 'chatter' example to perform a simple publisher program, and after setting up the communication everything works fine except for the topic 'chatter' it's not being seen even when using 'rostopic list'. can you please tell why this is happening? 
thanks in advance

yoneken, 2020-03-30:
Hi,
You have to launch [_rosserial_python_](http://wiki.ros.org/rosserial_python?distro=melodic) to subscribe to topics from embedded devices.

Abo3abed, 2020-03-30:
Thank you for your time!
But when I run rosserial_python serial_node.py _port:= _baud:= it gives me an error!
[ERROR] [1585574845.599400]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

5730289021-NN, 2020-04-28:
I used to get that error and I fixed that by matching baudrate between `_baud:=` and USART2 baudrate

yoneken, 2020-05-02:
Thank you for your helpful advice!
Yes. You have to direct your actual device. something like,

``
rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=38400
``

Actually, the default baudrate of rosserial is 38400.