yoneken/rosserial_stm32 #33: [ERROR] [1609744879.430792]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

deepchand41, 2021-01-04:
Using STM32F030 R8 Nucleo Board 
Using Ubuntu 20.04, ROS Noetic

Wayne-DWA, 2022-01-13:
@tahakenar @deepchand41 
check the DMA settings, set the mode to normal and then have a try.
cheers!
[image]

ImadUddien9543, 2022-12-31:
Hello, recently i was working on a college project that requires me to working on rosserial, the borad that i'm working on is stm32 nucleo f429zi
i configured usart3 (because i read on manuals that st-link on this board is connected to usart3, cmiiw) as DMA an RX mode is circular and TX mode is normal. Is what i'm doing correct? Because i'm constantly getting error (unable to sync etc.) 
Does the nucleo f429 need usb to ttl for rosserial or does the st-link usb port already suffice?
I tried to enlrage the buffer to 1024, and  other attempts like configure MCU settings in STM32CubeIDE (well, i searched for many youtube videos and other github repos hoping to solve this problem, but it hasn't yield any result)
Please let me know what caused this issue

[image]

[image]
[](url)

Sincerely, looking forward to your answers
Thank you
@yoneken @Wayne-DWA

JinK2000, 2023-05-09:
> @tahakenar @deepchand41 check the DMA settings, set the mode to normal and then have a try. cheers! ![cfg](https://user-images.githubusercontent.com/90710539/149395538-a66dc40f-1d64-4bc6-bfe3-13ffe66bf438.pn

it seems doesn't work :STM32F407+ubuntu20.04(notetic)

- Only in rare cases does it work：

[INFO] [1683649672.768868]: Requesting topics...
[INFO] [1683649673.098597]: Note: publish buffer size is 512 bytes

- Most of the time it reports an error like this and i don't know how to solve:

[INFO] [1683650864.691406]: Requesting topics...
[ERROR] [1683650879.695435]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

Hbbbbbby, 2024-09-20:
I encounter the same issues. Has it been solved now? I found that when the server sent a clock syn signal to the stm32 node, the usart of the node would be interrupted, causing that the message transmits incompletely which might be root of this problem. 

But I have no any clear train of thought about it.