yoneken/rosserial_stm32 #42: Type mismatch error in STM32Hardware.h

tianyuZ, 2022-09-23:
https://github.com/yoneken/rosserial_stm32/blob/7ccaf5befb5c717982bec8659ce37ddf33b89b58/src/ros_lib/STM32Hardware.h#L107
Type mismatch,should be modified to HAL_UART_Transmit_DMA(huart, tbuf, twind);

tianyuZ, 2022-09-23:
After looking at it in detail,The second half of message in case of tx bufferoverflow will not be transferred to the serial port through DMA because the first part of the buffer has not been transferred, so the previous logic is correct.

jaykorea, 2022-09-28:
Hello, I've read your reply about that problem...

That problem came to me and It occurs compile error.. 

Can I fix the line to this 
```
HAL_UART_Transmit_DMA(huart, tbuf, twind);
```
?? Waiting for your apply, thanks

maxi-naeher, 2022-09-28:
> After looking at it in detail,The second half of message in case of tx bufferoverflow will not be transferred to the serial port through DMA because the first part of the buffer has not been transferred, so the previous logic is correct.

What do you mean with previous logic?
I suggested the second DMA transfer because of this problem -> https://github.com/yoneken/rosserial_stm32/pull/40

jaykorea, 2022-09-28:
> What do you mean with previous logic?
>
> I suggested the second DMA transfer because of this problem -> https://github.com/yoneken/rosserial_stm32/pull/40

Thanks for your information.
Simply I have a issue that I cannot compile that line. 
Compile error occurs in stm32f466re board, Firmware version 1.24.2

maxi-naeher, 2022-09-28:
> > > After looking at it in detail,The second half of message in case of tx bufferoverflow will not be transferred to the serial port through DMA because the first part of the buffer has not been transferred, so the previous logic is correct.
> > 
> > 
> > What do you mean with previous logic?
> > I suggested the second DMA transfer because of this problem -> #40
> 
> Thanks for your information. Simply I have a issue that I cannot compile that line.(mentioned on issue #43) Compile error occurs in stm32f466re board, Firmware version 1.24.2

For me with Arm Compiler 6.18 it works fine. But obviously not for yours. U can just try the fix as suggested by @tianyuZ and test it.
In my opinion it should work, but I can't test it right know.

maxi-naeher, 2022-09-29:
> Thanks for your information. Simply I have a issue that I cannot compile that line.(mentioned on issue #43) Compile error occurs in stm32f466re board, Firmware version 1.24.2

Since @tianyuZ discovered a problem with the DMA transfer logic, you probably should use rosserial_stm32 from this commit:
https://github.com/yoneken/rosserial_stm32/commit/0d1c48b9427be8dec439065864b0b44c1fdb5ebd

tianyuZ, 2022-11-07:
> 

> Hello, I've read your reply about that problem...
> 
> That problem came to me and It occurs compile error..
> 
> Can I fix the line to this
> 
> ```
> HAL_UART_Transmit_DMA(huart, tbuf, twind);
> ```
> 
> ?? Waiting for your reply, thanks

Sorry, just saw your comment now, your fix is correct.

tianyuZ, 2022-11-08:
> What do you mean with previous logic? I suggested the second DMA transfer because of this problem -> #40

When the DMA transfer is completed, it will enter the interrupt and finally call the HAL_UART_TxCpltCallback callback function to transfer the remaining data again, so there is no need to immediately send out the second half of the message in the front of the tx buffer after a buffer overflow occured.