yoneken/rosserial_stm32 #45: how to add another STM32 Board

peaky-quest, 2023-04-15:
#define STM32F3xx  // Change for your device
#ifdef STM32F3xx
#include "stm32f3xx_hal.h"
#include "stm32f3xx_hal_uart.h"
#endif /* STM32F3xx */
#ifdef STM32F4xx
#include "stm32f4xx_hal.h"
#include "stm32f4xx_hal_uart.h"
#endif /* STM32F4xx */
#ifdef STM32F7xx
#include "stm32f7xx_hal.h"
#include "stm32f7xx_hal_uart.h"
#endif /* STM32F7xx */

This a piece of code from STM32Hardware header file, how to add stm32f103 microcontroller ? please guide

peaky-quest, 2023-04-15:
I am using CubeIDE

mink007, 2024-03-06:
I am assuming you have generated the boiler plate code for your STM32 project in STM32CubeIDE.

In the STM32Hardware.h simply add these lines:

#ifdef STM32F1xx
#include "stm32f1xx_hal.h"
#include "stm32f1xx_hal_uart.h"
#endif 

remove this line "#define STM32F3xx // Change for your device"