yoneken/rosserial_stm32 #47: wrong checksum for topic id and msg

corvusoul, 2024-02-23:
Anytime I publish anything I keep getting "[INFO] [timestamp]: wrong checksum for topic id and msg"

I have messed around with delays in code and baud rates and versions of ROS but of no use. I know it's not an error but it is annoying and it would be nice if I can fix this.

yoneken, 2024-02-28:
> "[INFO] [timestamp]: wrong checksum for topic id and msg"

This message often appears when messages are sent frequently, so could you try adding a wait after sending messages?
The problem may also be resolved by increasing the buffer size.

bdking3, 2024-04-12:
I am having the same issue and have verified that buffer size and delay size does not change the behavior. HAL_Delay of 1 or 1000 and buffer size of 512 or 2048, the issue remains.
In fact, the check sum failure occurs for the same number of packets regardless of loop time. If you poll faster, it occurs faster, if you poll slower, it happens slower. In both cases, the same number of packets go by before the first checksum fails. This then happens rhythmically forever. I am running on an STM32L476VGT6U.